LuxuSam/PhysicalButton #22: Ideas: Pi Shutdown and Double Click Input

jordankooyman, 2021-08-03:
It would be really cool if one of the button actions could be to safe shut down the Pi. I have tried a few other plugins, but they do not have the configurability I have seen here. It would also be cool if you could implement a double click input, so that if a button gets bumped, it won't necessarily trigger a function, particularly the Pi shut down.

LuxuSam, 2021-08-03:
Good idea, I had already thought about adding the possibility to run scripts or commands to the executable activities, that would then also include to shut the Pi down :D

Regarding your suggestion for the double click, _Hold Time_ in the button configuration is meant to keep the button from accidentally being triggered. If for instance you set the hold time to 1 second, the button has to be pressed 1 second in order to execute the activities, else the button press will be ignored.
I thought this was the easiest way to ensure that a button is only  triggered when you want it to.

closed 2021-08-06 by LuxuSam

LuxuSam, 2021-08-07:
@jordankooyman
With the new release you can add a system command for instance to shut down the pi :) Just add a system action and copy the shutdown command from _Settings > Server > Shutdown system_ and paste it in the input field. :D

Cheers,
Sam

jordankooyman, 2021-08-07:
Thank you very much! I hope this will work better than what I have tried in the past.

electronicm, 2021-09-01:
@jordankooyman Take a look at [dtoverlay:gpio-shutdown](https://www.google.com/search?q=dtoverlay:gpio-shutdown).
It is the native way to shutdown the Pi via gpio.
If you use the standard pin GPIO3, the PI can also be started with the button