LuxuSam/PhysicalButton #24: pause/resume action

tcurdt, 2021-08-08:
instead of having separate buttons for pause and resume it would be nice to also have a single button that toggles the printer state.

LuxuSam, 2021-08-08:
I like that idea! I did not think about that. I'll see what I can do in the next few days.
Cheers,
Sam

LuxuSam, 2021-08-08:
Hi, 
I added the toggle pause action to a feature branch, but I do not have a printer here at the moment to test if it works as supposed. The action was actually already given in the documentation of OctoPrint, but I missed it when I was looking at what actions to include.

 If you are interested in trying it out, you can install it with the following link:
`https://github.com/LuxuSam/PhysicalButton/archive/refs/heads/feature-pause-resume-action.zip`.
(Copy the link, go to _Settings > Plugin Manager > + Get More_, paste the link and hit install)

Cheers,
Sam

LuxuSam, 2021-08-11:
Hi,
Did you have time to test the toggle pause action? Does it work as it should?
If yes let me know, I can then include this in the next release :)

Cheers,
Sam

tcurdt, 2021-08-11:
I didn't get around to it yet.

How would that affect the currently installed plugin and its config? And how would I go back to the official release?

LuxuSam, 2021-08-11:
As the plugin has the exact same identifier, the official one would be uninstalled and the feature one would be installed. The config remains as the pi "thinks" it's an update for the plugin. The config still has the same structure as before. As far as I know, when you reinstall the official release, the action having the trigger pause would be set to none. But to be sure you could also remove that action from the button.

To revert to the official release you can either wait for me to release a new version, or you can follow the steps from before:
Install it with the following link:
[https://github.com/LuxuSam/PhysicalButton/archive/master.zip
](https://github.com/LuxuSam/PhysicalButton/archive/master.zip)
(Copy the link, go to Settings > Plugin Manager > + Get More, paste the link and hit install)

Cheers,
Sam

tcurdt, 2021-08-11:
I'll give that a try on the next print.

tcurdt, 2021-08-11:
works! :)

closed 2021-08-12 by LuxuSam