LuxuSam/PhysicalButton #26: GPIO buttons not working

calebmah, 2021-08-09:
Hi, first of all thanks so much for creating this plugin! I've previously been using the [enclosure plugin](https://plugins.octoprint.org/plugins/enclosure/) to control octoprint with physical buttons, but this plugin looks cleaner and easier to use!

I attempted to set this up but unfortunately was unable to get any of the buttons working. In my logs I see the below error:

```
2021-08-10 00:02:29,542 - octoprint.plugins.physicalbutton - DEBUG - Buttons to remove: []
2021-08-10 00:02:29,542 - octoprint - ERROR - Error while trying to migrate settings for plugin physicalbutton, ignoring it
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/__init__.py", line 512, in init_settings_plugin_config_migration_and_cleanup
    implementation._identifier, implementation
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/__init__.py", line 503, in settings_plugin_config_migration_and_cleanup
    implementation.on_settings_cleanup()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_physicalbutton/__init__.py", line 155, in on_settings_cleanup
    octoprint.plugin.SettingsPlugin.on_settings_cleanup()
TypeError: on_settings_cleanup() missing 1 required positional argument: 'self'

```

A little further down, I also this which seem to suggest that everything was set up ok:
```
2021-08-10 00:02:44,001 - octoprint.plugins.physicalbutton - DEBUG - Setting up buttons ...
2021-08-10 00:02:44,806 - octoprint.plugins.physicalbutton - DEBUG - Added Buttons: [<gpiozero.Button object on pin GPIO24, pull_up=True, is_active=False>, <gpiozero.Button object on pin GPIO23, pull_up=True, is_active=False>, <gpiozero.Button object on pin GPIO22, pull_up=True, is_active=False>, <gpiozero.Button object on pin GPIO27, pull_up=True, is_active=False>, <gpiozero.Button object on pin GPIO17, pull_up=True, is_active=False>]
2021-08-10 00:02:44,807 - octoprint.plugins.physicalbutton - INFO - Buttons have been set up!
```

Unfortunately, none of the buttons work when I press them. I've tried changing the mode from normally open to normally close as well but the buttons still do not react. The hold time is set to 100ms for all the buttons, if it matters. 

Is there anything that I have set up incorrectly, or is there any other way to troubleshoot this? Thanks for your help!

LuxuSam, 2021-08-11:
Hi,
First of all, thank your very much! I appreciate hearing that! It was my goal to have a very clear and easy use interface 😄 

Also thank you very much for reporting your error. I could not find a way to trigger the cleanup event. There seems to be a bug, in the cleanup function. 
I will fix that and hope that this also fixes your problem with the button setup. Especially as the _Debug log_ shows that the buttons have been set up correctly 😅. If this does not work, you could try to ssh into your pi and delete the physicalbutton entry in the _config.yaml_. But then you have to setup your button all over again.

Cheers,
Sam

LuxuSam, 2021-08-12:
Hi,
Sorry for not getting to this earlier 😅 I had the PR ready but did not have time to see if nothing broke until today.
Can you try the following:

`https://github.com/LuxuSam/PhysicalButton/archive/refs/heads/bugfix-cleanup-error.zip`
Copy the link, go to Settings > Plugin Manager > + Get More, paste the link and hit install.

Tell me if that works, as I am not able to reproduce that behavior.
If it does, I can include the fix in the next release.

Cheers,
Sam

calebmah, 2021-08-16:
Hey, thanks for the quick response! Apologies for the delayed response, I didn't realize you had replied already!

I installed the updated version and can confirm that the error with `on_settings_cleanup()` is fixed and I no longer see that in my logs. It turns out that this wasn't the cause of the plugin not working though. In actual fact, I had wired my buttons incorrectly and wired them all to 3v3 power instead of to ground! My buttons were wired following the instructions in [this project](https://www.thingiverse.com/thing:4024711) and I carelessly assumed that the wiring was the same. Changing the 3v3 wire to ground fixed the issues and my buttons are working as intended! 

A small suggestion I have would be to update the [plugin repository page](https://plugins.octoprint.org/plugins/physicalbutton/) with all of the same information as on the README. I originally missed the note on the wiring cause I was referring to the plugin repository page. 

Otherwise, thanks again for fixing this issue and for this fantastic plugin! I'm especially happy with the new pause toggle action in the latest release, I've been looking for a way to do exactly that. Speaking of which, would it be possible to add a similar start/cancel toggle action as well? :)

LuxuSam, 2021-08-17:
Hi,

Okay, nice, then I can release the fix for that bug :)

The plugin repository page also includes the information on how to wire the buttons. It Is written under the explanation for the GPIO pin :)
In addition, if you click on the (?) beside GPIO when you are configuring a button, you also get the information on how to wire the buttons.

Regarding your feature request, could you open a new issue for that, so I can close this one?

Cheers,
Sam

closed 2021-08-17 by LuxuSam