LuxuSam/PhysicalButton #27: clarify the user commands are run as

tcurdt, 2021-08-10:
While user `pi` can successfully run `sudo halt` from the terminal,
but this fails when called as system action from the plugin.

What's the best course of action to debug this?

LuxuSam, 2021-08-11:
Hi,
Is this the only command that does not work or are there also other commands?
I've read that the `sudo halt` command can have problems compared to the `sudo shutdown -h now` command.

Cheers,
Sam

tcurdt, 2021-08-11:
Can you first clarify - the commands get run as which user - `pi`?

I am surprised to hear about such problems. IIUC
[`shutdown` should just schedule `halt` IIUC](https://unix.stackexchange.com/questions/8690/what-is-the-difference-between-halt-and-shutdown-commands).

But I will give `sudo shutdown -h now` a try and report back.

LuxuSam, 2021-08-11:
I can't say with 100% certainty, but I think the OctoPrint host is run from the root user. And as far as I understand, the implementation for system commands spawns a child process from that user. 
So I believe it's the root user that runs the commands. 
I'll have a look into how to specify what user should run the command.

LuxuSam, 2021-08-11:
@skobkars Do you know the exact answer to this? :)

tcurdt, 2021-08-11:
```
octoprint.log.2021-08-10:2021-08-10 02:22:13,570 - octoprint.plugins.physicalbutton - INFO - Executing system command 'sudo halt'
octoprint.log.2021-08-10:2021-08-10 02:22:13,732 - octoprint.plugins.physicalbutton - ERROR - Error [1] executing command 'sudo halt': sudo: no tty present and no askpass program specified
```

It seems it's the lack of the tty that is causing the problem.
And octoprint is running as `pi`.

```
pi         299 10.7  6.3 225364 56888 ?        Ssl  11:43   0:53 /home/pi/oprint/bin/python3 /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000
```

tcurdt, 2021-08-11:
My bad - my `sudoers` file was still lacking the NOPASSWD in the right spot.

But I still would suggest to add the information of the user running the command to the UI somewhere.

skobkars, 2021-08-11:
Hi Torsten,

The system commands are run by the same user that runs OctoPrint, which is
the case for any plugin. So, I think we would only need to put that
information in the description or in the UI in case we
intentionally changed that for some reason. Which we didn't.

The logs specifically said that no askpass program was specified, which is
a great clue to what was happening.

Cheers,

skobkars, 2021-08-11:
> That's the developers POV. Do you honestly believe that all the people out
> there using octopi have a clue about these details? It's not that all that
> many plugins (in the grand scheme of things) actually run user defined
> system commands. So most user won't be aware.
> In retrospect this all was pretty straight forward - rather obvious
> really. Just my misconfigured sudoers file tripped me up. But if you care
> about the UX I don't think adding a "the commands are run as the user pi by
> default" would be such a bad thing.
> Just my 2 cents. Feel free to close if you disagree.

And here is the problem! I don't really know which user is running the
OctoPrint on any particular device. Therefore, I cannot really say that the
system command will be run as user pi, and will have to actually find it
out before saying so.

I guess we could say it is run by the same user that is running OctoPrint,
but in my view this is obvious and only needs to be specified if changed.

Maybe we can add that to the plugin webpage / documentation? Something like
this:

*System:*

*You can input any system command for your Octoprint host. NOTE: The
command or script will be executed by the same user that runs OctoPrint.*

Will that help you think?

On the other hand, I'm not sure if many users read these docs.

tcurdt, 2021-08-11:
> And here is the problem! I don't really know which user is running the OctoPrint on any particular device. Therefore, I cannot really say that the system command will be run as user pi, and will have to actually find it out before saying so.

Not sure I see that as a problem. Having the plugin finding the current user isn't particular hard,

```pwd.getpwuid(os.getuid()).pw_name```

but I think even that might be unnecessary.
I am pretty sure it's more than 90% of the user that will run OctoPrint on a pi anyway.

If you want to be correct and cover the rest you could also just something along the lines of "which user depends on your system, but most likely it will be pi". Or use the above code snippet.

> I guess we could say it is run by the same user that is running OctoPrint,
but in my view this is obvious and only needs to be specified if changed.

Really not to point fingers(!) but just to point out the relativity of "obvious": Even @LuxuSam wasn't totally sure "...I can't say with 100% certainty, but I think the OctoPrint host is run from the root user." and you expect that from your average 3d printing enthusiast?

In my experience when communicating across knowledge levels it's best to be explicit as possible.

> Maybe we can add that to the plugin webpage / documentation?

I agree, only people really searching will find it - but even in the project README would be a start.
But I personally would just add it as a note in the configuration section for commands.

Or if you want to be fancy - use the above snippet to show the current user.

skobkars, 2021-08-11:
Now you are talking like a non-programmer, lol. "More than 90%" rule is a
recipe for disaster. Programmers/coders are always thinking about "that one
guy", so if I offer any information it must be correct in *all cases*. I
don't want any more emails like yours. No offense.

On the other hand, if codes are up for the UI design it won't be any good
either. There will be 1,000 buttons that are barely used and tons of
cryptic details. I'm pretty sure that 90% of the 3d enthusiasts will not
even understand why I put that info there, or why it is important.

I'd say that putting that information in the documentation is a good
compromise. No need to clatter the screen with something that people only
need to find out once. Or zero times if they are in 10%.

*BTW, pi user in in sudoers file for shutdown command by default. It's just
that you used halt instead. If it was shutdown, you wouldn't be emailing us
at all.*

Regards,

tcurdt, 2021-08-11:
> Now you are talking like a non-programmer, lol. 

That's called empathy ;)

> "More than 90%" rule is a recipe for disaster. Programmers/coders are always thinking about "that one guy", so if I offer any information it must be correct in *all cases*. I don't want any more emails like yours. No offense.

Sorry, but knowing about the 80/20 rule is part of being a (good) programmer. No, it hasn't to be correct in all cases.
But hey - I even provided the code for it to be correct in all cases - and it's a one-liner.
And technically - even the sentence I provided is correct all the time.

> On the other hand, if codes are up for the UI design it won't be any good either. There will be 1,000 buttons that are barely used and tons of cryptic details. 

Not every programmer sucks at UI design. 80% might ;)

> I'm pretty sure that 90% of the 3d enthusiasts will not even understand why I put that info there, or why it is important. I'd say that putting that information in the documentation is a good [...]

Providing a hint does not have to clutter the screen.

And even the 10% would have to make some assumptions (or dig into octoprint) as there is such a thing as privilege separation.

But anyway - I don't think there is a point to argue. So I'll leave it at that.

 > BTW, pi user in in sudoers file for shutdown command by default. It's just that you used halt instead.

Do you mean the pi user is in the sudo group by default maybe?
I just did a fresh install. My sudoers file did not have a NOPASSWD entry at all.

 > If it was shutdown, you wouldn't be emailing us at all.

I am not quite sure I am following.
See above.

And I didn't email you. I opened an issue. I guess you got the email because you were tagged (or subscribed to all issues). You can just unsubscribe. Feel free to disengage at any time.

In the end I was just here to make the UX easier for the plugin users.

skobkars, 2021-08-11:
Hmm... I can't be 100% about the sudo now, as my system is not freshly
installed .

But, why do you think you can Shutdown System in OctoPi's menu?

pi user can use shutdown command without a password by default.

But not halt.

skobkars, 2021-08-11:
I created pull request for Sam. It has updated information page, talking
about sudo and user permissions.

Still don't think that it has to be in the UI. But you can always make your
own changes and request a pull too if you think this is necessary.

Regards

LuxuSam, 2021-08-12:
I agree with @skobkars. Having it in the documentation should be more than enough, else it would clutter the interface. I just added short tooltips to the interface, for more information it's better to have a look at the configuration part in the documentation.

Cheers,
Sam

closed 2021-08-12 by LuxuSam