LuxuSam/PhysicalButton #31: Requests Function: Acoustic Feedback

Fagge40, 2021-08-14:
I want to connect a buzzer to a GPIO to beep when a button is pressed, as an acoustic feedback to the buttons.

LuxuSam, 2021-08-15:
Hi,
Do you mean something like an activity to generate a HIGH output to a GPIO pin for a set amount of time?

I like the idea :)
I'll see how I can include it reasonably.

Cheers,
Sam

Fagge40, 2021-08-15:
Exactly :)

calebmah, 2021-08-16:
This is a good feature request, but just sharing a possible alternative way to achieve a similar result depending on your printer board and firmware. 

I use the GCODE command `M300 S100 P20` (https://marlinfw.org/docs/gcode/M300.html) to play a short beep whenever a button is pressed. Since this plugin allows listing multiple activities per button press, I just put this GCODE activity as the first in the list and its been working well!

Fagge40, 2021-08-17:
Ahh thanks. A good and simple solution to the problem. It works for me :)

LuxuSam, 2021-08-17:
Good idea @calebmah! 
As my printer (Prusa Mini) does not support the M300 command yet, I'll still keep this feature request in mind!

electronicm, 2021-09-08:
@LuxuSam
Does it make sense to leave this plugin exclusively for inputs and implement the controlling of outputs in a new plugin?
My request #43 would provide a good interface for this.

This way both plugins would remain as simple as possible and universal.

LuxuSam, 2021-09-09:
I had some time to think about how to add an output activity to my plugin and I think I came up with a pretty good idea which does not increase the complexity of the interface! 

I'll probably start implementing it in the next few days :D

closed 2021-10-11 by LuxuSam