LuxuSam/PhysicalButton #32: Multi function button

EugeneSlabbert, 2021-08-15:
First of all...great work, very convenient and easy to implement. Thank you.
I have a single (NO) button installed and set to shut down system when pressed and held for 2 sec. Is there any way to start up the Pi again with the same button? and secondly will it be possible to set the button to 500sm, 1000ms, and 2000ms to perform different tasks on hold time? for instance 500ms = Toggle pause, 1000ms = Cancel Print, and 2000ms = Shutdown system.

Cheers!!

LuxuSam, 2021-08-16:
Hi,
Thank you very much :D

In #23 GPIO3 was added. I was told that you can use that GPIO to also wake up the raspberry pi, but did not test that myself. So theoretically, you could setup a button which is connected to GPIO3 and use a system command to shut down the pi and use the same button to turn it on again.

Regarding your feature request. I think it could be doable, but a single button with no feedback whatsoever makes this very hard as you would have to set up intervals which you have to hit exactly, if not the wrong activity could be executed 😅 . So generally it could be done, but I don't think this is very user friendly and that would defeat the purpose of easy to use buttons. If you're scared to be running out of Ground pins, you should be able to use multiple buttons on one Ground pin without a problem :)

Cheers,
Sam

LuxuSam, 2021-08-29:
I'll think about how I could add it without breaking the current setup.

cbrherms, 2022-01-04:
I would like to third this, for the same reason as @mtabo - press to trigger an on action and hold for a few seconds to trigger an off action.

jneilliii, 2022-02-11:
you guys could use the toggle command instead. Using a system action with the following command.

```
/home/pi/oprint/bin/octoprint plugins psucontrol:toggle
```

LuxuSam, 2022-03-18:
I have an idea on how to implement this feature properly.
I'll start with it as soon as I have time!

coolcheat, 2022-12-17:
Newcomer here.

Love the plugin! would also love the ability to bind different activities depending on the hold duration.

Thanks!

stripu, 2024-05-21:
Hello! Any news on this? It would be a very helpful implementation for special situations. 
Thank you for your work on this plugin!