LuxuSam/PhysicalButton #33: Run full program

Pfeifdog, 2021-08-16:
Hi. 
Thank you for your work on this plugin. I was wondering if it would be possible, to load an entire program? I'm using this on a sandtable, and I want it to work offline. I am able to use the current plugin to home the machine. I've also managed to cheat a little, by copy and pasting the g code directly into the g code command. However, I was unable to paste a large g code file. I was successful, up to about 1,600 lines of code. The file I would like to run is around 12,000 lines of code. Any help on this, would be greatly appreciated.

Thanks, Mike Pfeiffer

LuxuSam, 2021-08-17:
Hi,
Thank you very much!

What do you mean by program? Do you want to run a gcode file with a button?

Cheers,
Sam

Pfeifdog, 2021-08-17:
Hi Sam,
  Yes. I’m trying torun a gcode file by pressing a button. My gcode file
would be about 10,000 lines of x and y positions.

      Thanks Again, Mike Pfeiffer

LuxuSam, 2021-08-17:
Okay nice,

I already have an idea how to include this more or less smoothly.

Cheers,
Sam

Pfeifdog, 2021-08-17:
Great, thanks!

LuxuSam, 2021-08-20:
Hi @Pfeifdog ,

The feature is now usable 😄 
If you are interested in trying it out already you can do so by completing the following steps:
1. Copy the following link: 
`https://github.com/LuxuSam/PhysicalButton/archive/feature-activity-select-file.zip`
2. Go to _Settings > Plugin Manager > + Get More_
3. Paste the link there
4. Hit install

I added an information link (the (i) button) to the new 'file activity'. Could you please check if what I wrote makes sense to you as a user?

If you are happy with the feature, I will add it to the next release!

Cheers and thank you very much for the support,
Sam

Pfeifdog, 2021-08-20:
Awesome, thanks! I’ll check it out later today, and let you know how it
goes.

Pfeifdog, 2021-08-20:
Hi Sam,

 The plugin update seems to be working. As far as the file format. I
wasn't sure what path to enter? So I just entered the file name as it
appeared in my octoprint upload section. For example (Spiral.gcode) and
added the run command after the file. When I pushed the button, the program
started running! If I have any other issues or questions, I'll let you
know. Thanks again for your work on this! You're Awesome!!!

Mike Pfeiffer

LuxuSam, 2021-08-20:
Hi,

Great to hear! 
Okay, then I'll have to rework the description a bit :)

A quick explanation regarding the path:
If the file would be situated in a folder named `Objects` you would have to put `Objects/Spiral.gcode` as a path.
I'll try to add that comprehensively to the information window. Thanks for the feedback!

Cheers!

Pfeifdog, 2021-08-20:
Thanks again! Just in case you’re interested? I’ve included a pic of what
I’ve been working on. It’s a project I found on Thingiverse.

LuxuSam, 2021-08-20:
I'm interested but unfortunately cannot see a picture? 😅

Pfeifdog, 2021-08-20:
I resent it? Maybe the file is too large?

LuxuSam, 2021-08-20:
Are you replying by email to the issue? 
I think you have to login into Github and upload it directly in a comment.

Pfeifdog, 2021-08-20:
[image]

LuxuSam, 2021-08-20:
Wow that looks very amazing 😄 Nice touch with the LEDs 😄

Pfeifdog, 2021-08-20:
https://user-images.githubusercontent.com/88604126/130293828-dbe8ebb4-fe69-43ff-b434-23c275a02b94.mp4

LuxuSam, 2021-08-20:
Wow, I‘m very happy that my plugin is used in such a nice way 😁 I‘m glad I could help!

LuxuSam, 2021-08-21:
Does this help window make more sense? 
[image]

Pfeifdog, 2021-08-21:
Yes, it does. Looks good to me 👍🏻

LuxuSam, 2021-08-21:
Thank you very much!

closed 2021-08-21 by LuxuSam