LuxuSam/PhysicalButton #34: [FEATURE] Start/Cancel toggle action

calebmah, 2021-08-17:
**Describe the solution you'd like**
It would be awesome if there were a Start/Cancel toggle action just like the recently added pause toggle action! That would allow starting a print (if nothing is printing) and cancelling an ongoing print with the same button! 

**Additional context**
Thanks again for the quick turnarounds and releases!

LuxuSam, 2021-08-17:
Hi,

Thanks for the request.
Do you mean like start the last selected file? Like the usual start action, but with the addition if you are already printing that the print is canceled?

Cheers,
Sam

closed 2021-08-18 by LuxuSam

LuxuSam, 2021-08-18:
I finished programming the new action, it will be added in the next release.