LuxuSam/PhysicalButton #37: Won't install on Octoprint 1.6.1

nifty1a, 2021-08-18:
Plugin doesn't appear in Plugin Manager, and I don't seem to be able to install plugin manually,
on Octoprint 1.6.1 on OctoPi 0.17.0

I think I saw a message scroll up about wrong Python versions

nifty1a, 2021-08-18:
It was hidden in Plugin Manager as it says it is incompatible, but no indication as to why, or how to fix/make my installation compatible!
[image]

LuxuSam, 2021-08-18:
Hi,

The only thing that I can imagine is that you are still using Python 2, but the plugin just supports Python 3 😅 

Cheers,
Sam

nifty1a, 2021-08-18:
Thanks, I have just found that I am a version behind on OctoPi and that is the one that migrates to Python 3... I will do the necessary and get it sorted, thank you