LuxuSam/PhysicalButton #39: [BUG] Short bug description

Guid-o-nator, 2021-08-20:
**Plugin Version**
unknown, currently available under https://github.com/LuxuSam/PhysicalButton/archive/master.zip

**Describe the bug**
- Cannot find PlugIn when searching in Plugin Manager
- Cannot install, wrong Python version:
- ERROR: Package 'Physical-Button' requires a different Python: 2.7.16 not in '>=3, <4'

**To Reproduce**
OctoPrint 1.6.1 - Python 2.7.16 - OctoPi 0.17.0

**Expected behavior**
- Find Plugin in Plugin Manager
- Install Plugin

LuxuSam, 2021-08-20:
Hi,

Thanks for the bug report, but there is nothing I can do about that.
I made the plugin support only Python 3. You seem to have Python 2 installed.
If you want to update Python on your raspberry pi, you can read through this [Upgrade your OctoPrint install to Python 3!](https://octoprint.org/blog/2020/09/10/upgrade-to-py3/)

Cheers,
Sam

rnsi, 2021-08-21:
I get the same error. I have python 2 and 3 installed, but system info shows "env.python.version: 2.7.16". Running OctoPrint 1.6.1. Maybe instead of calling python we need to call python3?

LuxuSam, 2021-08-21:
What is the output of `pip --version` and `python --version` on your pi?

LuxuSam, 2021-08-21:
You need to have Python 3 set as standard as I cannot influence how the plugin is installed 😅

rnsi, 2021-08-21:
Older versions of OctoPi used Python2 as the default. Use the script here: https://octoprint.org/blog/2020/09/10/upgrade-to-py3/ to upgrade the OctoPi installation and this plug-in will install. This *may* break old plug-ins that are Python2 only.

Setting the OS default Python version is not enough. You need to change OctoPi.

LuxuSam, 2021-08-21:
Okay, I wasn't completely sure about that. But I also posted that link already above :)

Cheers,
Sam

closed 2021-08-22 by LuxuSam