LuxuSam/PhysicalButton #43: [FEATURE] Actions from other plugins

electronicm, 2021-09-02:
I use your awesome plugin to switch my PSU on or off via button.
The whole thing is controlled by [OctoPrint-PSUControl](https://github.com/kantlivelong/OctoPrint-PSUControl).

Currently I have the fatal problem that the PSU can be switched off by a button while printing. (System: `gpio write X 0`)
Since for this scenario a mechanism is already integrated in the PSUControl, I will execute in the future in your plugin per system the command `~/oprint/bin/octoprint plugins psucontrol:toggle`.
However, it is unpleasant to have to take the CLI detour, which makes the process unnecessarily complex and time-consuming.

**Do you have the possibility to make all commands of installed plugins available under Actions as well?** (`~/oprint/bin/octoprint plugins`)
The goal should be a more user-friendly way to use all the possibilities of Octoprint and his plugins.

LuxuSam, 2021-09-05:
Hi, thank you very much :)

I like the idea to have action of other plugins included in my plugin! I'll have a look at if and  how this can be achieved!

Cheers,
Sam

electronicm, 2021-09-08:
@LuxuSam
Does it make sense to leave this plugin exclusively for inputs and implement the controlling of outputs in a new plugin?
My request #43 would provide a good interface for this.

This way both plugins would remain as simple as possible and universal.

LuxuSam, 2021-09-27:
I'm sorry but I cannot find out how to get the available commands from other plugins 😅  
If someone knows how to achieve this, I would be happy to get a hint on that or a PR :)

electronicm, 2021-09-27:
I'll see if there is any information on the weekend.

OllisGit, 2022-01-28:
Hi @LuxuSam, 
if the plugin author doesn't provide a public api (web or internal) to execute tasks or read data you can directly get the plugin implementation and call the function directly...but this is a risky thing.

I wrote a helper function in my PrintJobHistory-Plugin that delivers the 3rd party plugin instance. On that instance you can call the needed function.
See https://github.com/OllisGit/OctoPrint-PrintJobHistory/blob/1.16.0/octoprint_PrintJobHistory/__init__.py#L291-L332

Usage:
```
pluginInfo = self._getPluginInformation({ "key": "costestimation", "minVersion": "3.4.0"})
self._costEstimationPluginImplementationState = pluginInfo[0]
self._costEstimationPluginImplementation = pluginInfo[1]

..do some state checks (installed/disabled)

self._costEstimationPluginImplementation.getCostData()

```

If the 3rd party plugin is willing to provide an api, it should use the "helpers-feature"
https://docs.octoprint.org/en/master/plugins/helpers.html#helpers

Or depending of the "usecase" it could fire an custom-event and a listener-plugin can react in "realtime" instead of polling the information from the plugin.
See
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress#events

LuxuSam, 2022-02-01:
Hi @OllisGit,

Thank you for the clarification, provided documentation, and your code examples :D

I'll look into it in more detail as soon as I'll have time to work on the plugin again! Exams are coming up.

jneilliii, 2022-02-11:
For the original request, PSU Control supports CLI commands so you could use system and the command `/home/pi/oprint/bin/octoprint plugins psucontrol:toggle` I think. 

https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/CLI

kantlivelong, 2022-04-24:
There are helper functions available.

https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Developers

closed 2022-05-03 by LuxuSam