LuxuSam/PhysicalButton #52: [QUESTION] List of actions

alphanumericaltycoon, 2021-12-16:
**Plugin Version**
Physical Button: 0.5.0

**Question**
Hi. Sorry for my noob question but is there anywhere that I can read up on what the various 'actions' do? Are these octoprint commands/plugin commands or something else?

At the moment my plugin activities has action: start latest selected. Does this mean it will start/print the most recent/latest thing that was printed?

LuxuSam, 2021-12-16:
Hi,

I did not think about adding complete descriptions for the different custom actions. I should add descriptions to the plugin pages :)
Most actions are from Octoprint, but some are custom. 
Do you mean the 'start latest' action? That action starts your latest uploaded file.
When you just printed a file and want to reprint it, just use 'start', but that won't work after rebooting the pi or having the pi disconnected from the printer.

Cheers,
Sam

jneilliii, 2022-02-11:
This would probably be the comprehensive list of supported actions by OctoPrint, but you can also use the custom actions plugin and define your own as well. Not sure how this plugin would react to those. 

https://docs.octoprint.org/en/master/features/action_commands.html

closed 2022-02-23 by LuxuSam