LuxuSam/PhysicalButton #57: [QUESTION] Toggle light on/off

Stargrove1, 2022-02-20:
I am about to add some physical buttons and was wondering if it is possible to use one button only to toggle the lights. My initial thougt was, to use a short press and a long press, but I just read another comment, explaining that it is not possible. 

Is there a simple solution or is it not possible at all?

The lights I am using is a Neopixel attached to the spider board, it can be controlled by gcode commands executed by klipper

LuxuSam, 2022-02-20:
Hi,

No, sadly in the current state it is not possible to do it like that. I also don't think that a toggle gcode action would be useful in other cases than your specific use case.

Maybe I'll add the possibility of toggling between two different activities in the future. 

Cheers,
Sam

Stargrove1, 2022-02-22:
Hi Sam,

I have seen mutliple requests about distinguishing between the duration of a button press, this would also be a possible solution, at least to have different actions on a short press and long press (maybe with a configurable threshold).
A toggle function may even be less usefull, as the action could have been triggered by another command with no feedback to the plugin, meaning if the light was turned off by the ui and the next action would be to turn it off on button press.
Maybe you can reconsider implementing the feature to have multipurpose button press.

Benjamin

closed 2022-03-12 by Stargrove1