LuxuSam/PhysicalButton #62: Button action on Raspi startup

seppelicous, 2022-04-04:
**Plugin Version**
0.5.0

**Question**
I use GPIO17 to send an action. The button is connected to ground and it works as intended.
Problem is on raspy startup. I am not an Raspberry expert, but on on start the GPIO17 pin is low and that seems to activate the action once. How can I prevent this behaviour? Do I have to use an pul up resistor or a different pin to get rid of this?

My pins 1-8 are already in use, so I cannot use pins that are HIGH on startup.

Thank you!

LuxuSam, 2022-04-05:
Hi,

That is really strange and I cannot think of a reason why this should be happening. 
You should not have to use a pull-up as I set those internally. You could try another pin, but as already said this should not happen. Do you have it set up as "Normally Open" or "Normally Closed"?

I could have a look at your octoprint.log to see if I get a clue from there.  
For this go to the logging tab in your settings and add octoprint.plugins.physicalbutton with debug level selected.

Cheers,
Sam

seppelicous, 2022-04-05:
Thank you for your fast reply. It is "Normaly open". I changed the GPIO twice and now cannot reproduce it anymore. That's weird. Now, since was another issue, I will close this.

closed 2022-04-05 by seppelicous