LuxuSam/PhysicalButton #63: [QUESTION] Run a system command

seppelicous, 2022-04-05:
**Plugin Version**
0.5.0
**Question**
I can use "kasa --host 192.168.178.27 on" as a pi user to switch my Kara socket on. Using this as a system command in your plugin does not seem to work. Am doing it right? "sudo" did not help either.
Can I use any command that I use in terminal?

LuxuSam, 2022-04-06:
You should be able to use any command you can use in the terminal. Could you attach your `octoprint.log`? There should be an error message if something went wrong.

seppelicous, 2022-04-08:
Good idea with the log...

> 2022-04-08 11:26:26,473 - octoprint.plugins.physicalbutton - DEBUG - Reacting to button Steckdose - 8
2022-04-08 11:26:26,474 - octoprint.plugins.physicalbutton - DEBUG - Sending activity with identifier 'kasa --host 192.168.178.27 on' ...
2022-04-08 11:26:26,474 - octoprint.plugins.physicalbutton - INFO - Executing system command 'kasa --host 192.168.178.27 on'
2022-04-08 11:26:26,534 - octoprint.plugins.physicalbutton - ERROR - Error [127] executing command 'kasa --host 192.168.178.27 on': /bin/sh: 1: kasa: not found
2022-04-08 11:26:26,535 - octoprint.plugins.physicalbutton - ERROR - The activity with identifier 'kasa --host 192.168.178.27 on' failed! Aborting follwing activities!

I wonder why it cannot execute "kasa" as works flawlessly in terminal, also with user "pi" without "sudo".

LuxuSam, 2022-04-08:
I did a bit of research and it seems like the module I'm using is not using the bash shell but the sh shell. I'll have to inform myself on how to change that. Maybe that could resolve your issue.

LuxuSam, 2022-04-08:
Could you reinstall the plugin with this link
`https://github.com/LuxuSam/PhysicalButton/archive/refs/heads/bug-bash-shell.zip` 
to try it out?

If you do not know how to install it with a link do ass following:

1. Open **Settings**
2. Go to **Plugin Manager**
3. Click **Get More**
4. Paste the link under **... from URL**
5. Hit **Install** next to that text field

seppelicous, 2022-04-09:
Unfortunately it still throws an error, as also "bash" seems not to work.
`octoprint.plugins.physicalbutton - ERROR - Error [127] executing command 'kasa --host 192.168.178.27 on': /bin/bash: kasa: command not found
`

I wish I could help, but I have little to no experience with this.
Nevertheless I found a possible workaround with MQTT messages, which might be an interesting feature for this plugin too. 
The buttons send individual messages to an mqtt broker and the action is processed in NodeRed.

LuxuSam, 2022-04-09:
To test it out I also installed `python-kasa`, with `~/oprint/bin/pip install python-kasa`.
I am able the command with `/home/pi/oprint/bin/kasa --host 'local ip address' on` from a button 😅
So you'll probably also have to specify the absolute path to the command.

closed 2022-04-12 by seppelicous