LuxuSam/PhysicalButton #68: [FEATURE] Temporarily disable a button?

wdonders, 2022-06-16:
Fighting my EFI interference issues, I wanted to temporarily disable some buttons to prevent problems due to false triggers during printing. I ended up to completely delete the buttons, as there is no way to disable and enable a button via a checkbox?
Some buttons have more actions and multiple lines of gcode, which I had to save in a txt file for future use.
(Since there is also no Backup feature to save and restore your settings... that could be another nice feature though ;-))

So, an enable/disable checkbox for every button would be a nice addition I think.

LuxuSam, 2022-06-16:
You can set gpio pin to none, then it's ignored by the plugin :)

LuxuSam, 2022-06-16:
The backup feature is a good idea though :)

wdonders, 2022-06-16:
> in to none, then it's ignored by the plugin :)

Yes, you're right, but i have 10 buttons and then I have to write down the button/GPIO combination somewhere 🙄

LuxuSam, 2022-06-16:
That's true, having an option to disable buttons would probably make sense.

As a workaround, until I get to implementing the feature requests, you could add the pin number to the button name for instance and set it to none to disable it. That's still better than deleting the buttons.

LuxuSam, 2022-06-18:
I added a feature branch with the proposed changes, if you are interested in them out do as following:

If you do not know how to install it with a link do as following:

1. Open Settings
2. Go to Plugin Manager
3. Click +Get More
4. Paste `https://github.com/LuxuSam/PhysicalButton/archive/refs/heads/feature-additional-button-options.zip` under ... from URL
5. Hit Install next to that text field

wdonders, 2022-06-19:
Works great, nice big enable buttons clearly visible in the button list, very clever! Thanks a lot!