LuxuSam/PhysicalButton #72: [FEATURE] Add button on navbar

DevooUP96, 2022-08-01:
Hello i would like to ask if its possibile to add a button on navbar that are syncronized with real button status
This idea come when you want able to power on one Relay with Real Button or Via Octoprint Web
It would be nice to have the possibility to access the button through the Octoprint interface for example i pressed Real button "z" the Gpio go 1 and have on Octoprint Interface the button "z" also changed status and able to change status of the button to 0

LuxuSam, 2022-08-04:
Hi,
I'll think about it, but I kind of like the idea.
I'm having exams at the moment so I will not get time for some weeks.

Cheers,
Sam

DevooUP96, 2022-08-04:
Sure not problem Exams first!, i tried my self to dev this part of code but unfortunely im not good with jinja and python so hope you will dev when you have time, good luck for exams XD

svcru, 2025-01-16:
This is a very useful and necessary feature. Could you please let me know if this has been done? I would be glad if so.