LuxuSam/PhysicalButton #74: [BUG] GCODE button only works with a connected printer in the Octoprint interface

UlliMUC, 2022-12-06:
**Plugin Version**
Physical Button (0.6.1)

**Describe the bug**
Connected simple GCODE (Extruder Temp -> 190) to a Button.

That button only works, when octoprint is connected to the printer in the browser, not in the following state or completely without connecting octoprint.

**To Reproduce**
Power up raspberry, press a button with gcode -> nothing happens. 
Power up raspberry, connect the printer in the web surface of octoprint, press a button with gcode -> works fine. 

**Expected behavior**
I think that the plugin should also work completely "offline", right? the other buttons, which control relays, work without any problems - only the ones with a gcode don't.

**Screenshots / Logs**

[image]

[image]

**Additional context**
If applicable, add any other context about the problem here.

LuxuSam, 2022-12-06:
Hi,

I don't think I can do something about this as you cannot send Gcode to the printer when it's not connected to Octoprint. I hand the Gcode commands to Octoprint, which is then responsible to deliver them to the printer.
Or am I not understanding your problem? If no, please attach the octoprint.log file (with debug level enabled for physicalbutton) when you reproduce the bug.

Cheers,

UlliMUC, 2022-12-06:
Thanks for this quiek reply!

I interrupt the power supply to the printer via a relay. Is it possible for octoprint to automatically connect to the printer as soon as it has power? octoprint doesn't need the web interface to communicate with the printer, does it?

LuxuSam, 2022-12-06:
Octoprint does not need the web interface to be open.

Do you use my plugin to trigger the relay? If yes, put a connect action into the activity list behind the output activity.
If not, i'd say that you could try to place a connect activity just in front of the gcode. 

Maybe the gcode activity has to be delayed until the printer has fully connected. Then you could (as a workaround) put a synchronous output activity after the connect activity and select a pin of the raspi you are not using. You have to set the time to a bit more than start time of the printer and connect time.

UlliMUC, 2022-12-06:
Thanks for your hints!!

I'm trying to send the "connect" command 9 seconds after the printer relay closes. unfortunately this doesn't work as expected:
[image]

LuxuSam, 2022-12-06:
Yes, for the fake delay to work you have to specify a gpio pin that you are not using on your board. In my code the output activity is ignored when gpio is set to none.
But it would definitely make sense for me to add a delay activity for this use case. I bet there are other cases where this could be useful.

closed 2023-01-12 by LuxuSam