LuxuSam/PhysicalButton #79: [QUESTION] Sometimes save doesn't save changes

wanamaa, 2023-07-16:
**Plugin Version**
0.6.1

**Question**
Just getting started with your plugin.  I started by adding a button to Home on GPIO13, set the Action to home, and saved.  I realized later I wanted to just send the Gcode G28 instead.  Went in, changed it, hit save, and tried using the button.  I double-checked my settings for the button, and it didn't change the action from Home to G28 like I expected.  Note that the printer connection was established at the time.  I performed a restart, didn't reconnect my printer yet, adjusted the settings, and now they save.  Is this normal behavior for the plugin - to not be connected to the printer in order to make button changes?

**Additional context**
Add any other context about the question here.

LuxuSam, 2023-07-17:
Hi,
That is no usual behavior, the printer's connection state should not matter to saving the settings.
Did that only happen once? Maybe Octoprint had some problem?

If it happens again, I will investigate.
Try enabling debug logging for Physical Button in the Octoprint settings. That way I can better check if something goes wrong :)

Cheers,
Sam

wanamaa, 2023-07-18:
I think I figured it out.  I had another plugin called GPIO that I _THOUGHT_ did what your plugin did, but it tells GPIO to turn on/off via software buttons.  In my troubleshooting yesterday, I remove that plugin and rebooted, then I found I could have changes saved.  Confirmed today that I can make changes and save reliably now via multiple logged in devices (laptop and mobile).  Thanks!  If I have any more hiccups, I'll generate log files for you :)

Thanks,
Alex

LuxuSam, 2023-07-18:
Great to hear!

closed 2023-07-18 by LuxuSam