LuxuSam/PhysicalButton #82: Doesn't work on new installation

mikulik86, 2023-10-05:
Hi, I can't get the plugin to work on a new OctoPrint.

I installed new instance of Octoprint using Raspberry Pi Imager (Image OctoPi 1.0.0 with OctoPrint 1.9.2). What ever I do, the Physical Button is just not working. Even though other GPIO plugins do work (i.e. Octolight, PSU Control including GPIO sensing)

But when I use the old SD with the original installation (OctoPi 0.18.0 with OctoPrint 1.9.2) everything works fine including the Physical Button!

I doublechecked everything. All the settings are the same, the wiring is the same, even raspi-gpio registers the signals correctly in both cases. I'm totally desperate. Please, help me figure out what am I missing.

LuxuSam, 2023-10-11:
Hi,
Could you please attach the octoprint.log file? You can download it in the settings under OctoPrint/Logging by clicking on the download button on the right side of the file.

Maybe I can see where the problem comes from.

Cheers,
Sam

tomasbubela, 2023-10-30:
The same for me. Doesn't work  with OctoPrint: 1.9.3 .

`2023-10-30 09:46:19,037 - octoprint.startup - INFO - Starting OctoPrint 1.9.3
2023-10-30 09:46:19,038 - octoprint.startup - INFO - ******************************************************************************
2023-10-30 09:46:19,544 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2023-10-30 09:46:19,545 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2023-10-30 09:46:19,545 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2023-10-30 09:46:19,895 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2023-10-30 09:46:19,911 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2023-10-30 09:46:23,395 - octoprint.plugin.core - ERROR - Error loading plugin physicalbutton
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1294, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_physicalbutton/__init__.py", line 7, in <module>
    from .lib.gpio_setup import setup_buttons, remove_buttons, remove_outputs
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_physicalbutton/lib/gpio_setup.py", line 9, in <module>
    from gpiozero import Button, OutputDevice
  File "/home/pi/oprint/lib/python3.7/site-packages/gpiozero/__init__.py", line 19, in <module>
    from .pins import (
  File "/home/pi/oprint/lib/python3.7/site-packages/gpiozero/pins/__init__.py", line 21, in <module>
    from ..devices import Device
  File "/home/pi/oprint/lib/python3.7/site-packages/gpiozero/devices.py", line 19, in <module>
    from importlib.metadata import entry_points
ModuleNotFoundError: No module named 'importlib.metadata'
2023-10-30 09:46:23,861 - octoprint.plugin.core - INFO - Found 22 plugin(s) providing 21 mixin implementations, 40 hook handlers
2023-10-30 09:46:24,043 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2023-10-30 09:46:24,207 - octoprint.server - INFO - Intermediary server started
2023-10-30 09:46:24,256 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2023-10-30 09:46:24,290 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2023-10-30 09:46:24,298 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2023-10-30 09:46:24,298 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2023-10-30 09:46:24,299 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2023-10-30 09:46:24,300 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2023-10-30 09:46:24,300 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_read'), Need(method='role', value='plugin_announcements_manage')")
2023-10-30 09:46:24,301 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2023-10-30 09:46:24,302 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_GRANT (needs: "Need(method='role', value='plugin_appkeys_user')")
2023-10-30 09:46:24,302 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2023-10-30 09:46:24,303 - octoprint.server - INFO - Added new permission from plugin firmware_check:

LuxuSam, 2023-11-05:
Could you check what Python version you are running?

tomasbubela, 2023-11-06:
Python 3.7.3 
OctoPrint 1.9.3, OctoPi* 0.18.0 (build 2022.01.20.112925

sw4280, 2023-11-10:
Helllo
i have the same problem but with
Python 3.9.2
OctoPrint 1.9.3, OctoPi* 1.0.0 (build 2023.03.02.160739)

LuxuSam, 2023-11-12:
The only thing I can think of is to specify which version of the gpiozero library to use. 
Since I cannot recreate the problem on my side, could you try installing the physical button plugin using the following link:
`https://github.com/LuxuSam/PhysicalButton/archive/refs/heads/82-bug-on-new-installation.zip`
If you're unsure how to install it via the link, first click on +Get More in the Plugin Manager tab in OctoPrint. At the bottom of that window, you can paste the link and click install.

Cheers,
Sam

LuxuSam, 2023-11-13:
It should fix the problem for users with Python 3.7. 
I do not understand why Python 3.9 would not work. 
@sw4280 could you also attach your octoprint.log file?

sw4280, 2023-12-20:
Hi LuxuSam,
i`m sorry, this was my fault.
The installation worked without issue, but a configured button was not working.
The problem was i had also OctoLight installed which was not working in BCM mode.
I uninstalled OctoLight now and used OctoRelay instead, and now also Physical Buttons are working without any problem...

closed 2024-02-04 by LuxuSam