LuxuSam/PhysicalButton #85: Cannot Install - "Install Unknown"

antonyburden, 2023-11-12:
**Plugin Version**
0.6.1

**Describe the bug**
After install and prior to restart - this message appears "Install Unknown:"

Python 3.7.3
OctoPi 0.17.0

The plugin does not show up withing OctoPrint

LuxuSam, 2023-11-12:
Hi,

Could you add your octoprint.log file? Without that file, I cannot help you.

Cheers,
Sam

LuxuSam, 2023-11-12:
This bug could also be related to issue #82. Please check if my proposed solution there works for you.

antonyburden, 2023-11-12:
Thanks, your new install URL has worked. I've attached it anyway:
[octoprint.log](https://github.com/LuxuSam/PhysicalButton/files/13328422/octoprint.log)

LuxuSam, 2023-11-12:
Thanks! 
Yes, it is the exact same problem as in the other issue. I'll wait until I get some more feedback there before I publish an official release. When I update, you should receive a notification in the OctoPrint interface, and you can install it from there as usual.

I'll close this issue!

closed 2023-11-12 by LuxuSam