LuxuSam/PhysicalButton #89: Not Working After Install

mxirwin, 2024-05-14:
**Plugin Version**
0.6.2
OctoPrint 1.10.0
Python 3.9.2
OctoPi* 1.0.0 (build 2024.04.25.094544)

**Question**

This is for a sand table I am building: https://www.diymachines.co.uk/kinetic-sand-art-coffee-table-self-drawing

Fresh install of plugin, seems to be installed correctly but I can't seem to get it to work. I have a known good button connected to GPIO24 and ground. Just trying to test it by having it home the machine, but after making sure the machine is connected and homing works via OctoPrint, when I push the button nothing happens. I have also tried it on GPIO23.

I tried configuring the button two ways to see if that made a difference, but no luck.

Is there some setting I could have missed somewhere?

[image]
[image]
[image]
[image]

LuxuSam, 2024-05-17:
Hello,

Your configuration looks correct. Could you please enable debugging for the PhysicalButton Plugin in the Logging tab within your OctoPrint settings? Perhaps the output log will reveal what is going wrong.

Cheers,
Sam

mxirwin, 2024-05-17:
Here is the logfile after pushing the button several times.
[octoprint.log](https://github.com/LuxuSam/PhysicalButton/files/15349152/octoprint.log)

LuxuSam, 2024-05-18:
I used the gpiozero library to write this plugin. It seems like it failed to install properly. If you have already some configuration, you can save the configuration file with the buttons in the header in the plugin settings. 

Could you then try to uninstall the PhysicalButton plugin and afterward remove gpiozero manually? To remove gpiozero, you need to SSH into your Pi and run `pip uninstall gpiozero` or `pip3 uninstall gpiozero`. 

Afterward, reinstall the plugin and check if the buttons work. If they still do not work, please send me the log file once more :)

mxirwin, 2024-07-05:
Finally got a chance to get back to this project.

Here are the results when I tried to uninstall gpiozero via ssh with both
commands.

[image: image.png]

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

mxirwin, 2024-07-05:
Looks like the image may not have come through on my last email reply.

[image]

mxirwin, 2024-07-05:
This is the project I am working on:

https://www.diymachines.co.uk/kinetic-sand-art-coffee-table-self-drawing

In the comments someone else who built this sand table was able to use your plugin to make the sand table a standalone project so you didn't have to log in via phone or computer and manually run the table through octoprint.

Also just an FYI, when I SSH into the pi, I used the default user and pass of pi and raspberry, in case that makes a difference.

mxirwin, 2024-07-08:
Just another update. Other than OctoPrint and a playlist plugin, the pi didn't have anything or any special configuration. I wiped the SD card and reinstalled OctoPrint and the playlist plugin, and PhysicalButton. I configured it the same way as before and it still does not seem to work for me.

mxirwin, 2024-07-10:
I did some tests by installing a plugin called gpio control. This test verified that the physical button plugin was working correctly. Somehow after doing the test and uninstalling gpio control, my actual physical button started working via the physicalbutton plugin.

closed 2024-07-10 by mxirwin

LuxuSam, 2024-08-01:
Hi, sorry for the late reply, I'm super busy at the moment.

I'm happy to hear that you solved the problem yourself!
Can you tell me what python version you are running on your rapsberry pi?

Cheers,
Sam