LuxuSam/PhysicalButton #90: [QUESTION] Can't get this to work

rs-garrick, 2024-08-06:
**Plugin Version**
0.6.2

**Question**
I'm trying this plugin for the first time on octopi and a new raspberry pi 4.  Install and config seems fine but it's not responding to events. A small python script using `RPi.GPIO`, running as the same user, responds to the button push, so at least I know the basic hardware, OS, and file privs pieces are OK.

I can see that octoprint has a file descriptor for `/sys/devices/platform/soc/fe200000.gpio/gpiochip0/gpio/gpio17/value` open, so it seems to be mostly good. I've enabled debug logging and I see activity when saving configs in the plugin UI.

Octoprint doesn't respond to the button. No actions will execute. Nothing is logged. `strace` on the octoprint process doesn't show any activity.  I'm not sure how to debug from here.

Thanks in advance.

**Additional context**
Everything is pretty new.
Build 2024.06.18.085851 with "webcamd", based on OctoPi 1.0.0, running on Raspberry Pi 4 Model B Rev 1.5

LuxuSam, 2024-08-12:
Hi,
Could you check what version of python you are running? Additionally check the version of gpiozero. Maybe there is a version mismatch.

You can can do that by using the following commands in the terminal of the raspberry pi:
- ``` python3 -version ```
- ``` pip3 show gpiozero ```

Cheers,
Sam

rs-garrick, 2024-08-12:
It's just whatever came with octopi. I've not really done much to this system.
```
garrick@octoprint:~ $ python3 --version
Python 3.9.2
garrick@octoprint:~ $ pip3 show gpiozero
Name: gpiozero
Version: 1.6.2
Summary: A simple interface to GPIO devices with Raspberry Pi
Home-page: https://gpiozero.readthedocs.io/
Author: Ben Nuttall
Author-email: ben@bennuttall.com
License: BSD-3-Clause
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
```

LuxuSam, 2024-08-13:
Thank you! 
This confirms my suspicion. My setup logic installed the wrong gpiozero version for your Python version. I'll try to fix this.

LuxuSam, 2024-08-13:
Hi, could you try installing the physical button plugin using the following link:
```https://github.com/LuxuSam/PhysicalButton/archive/refs/heads/82-bug-on-new-installation.zip```
If you're unsure how to install it via the link, first click on +Get More in the Plugin Manager tab in OctoPrint. At the bottom of that window, you can paste the link and click install.

Cheers,
Sam

rs-garrick, 2024-08-15:
So this is weird.... it worked... once?  My test.sh system action was executed once.  I've not gotten it to execute from the button press a second time.

LuxuSam, 2024-08-16:
Could you please attach the `octoprint.log` file?

To do this, enable debugging for the plugin in the settings under **OctoPrint/Logging**, then restart OctoPrint and try pressing a few buttons.

After that, you can download the log file in the logging settings by clicking the download button next to the file.

rs-garrick, 2024-08-19:
Just FYI, I'm out of town now.  I won't be back till next week.

BoilingFrog73, 2025-03-29:
Hi. I'm also encountering non functionality of Physical Button plugin after install. I'm on a RPi 3a+ so I think I'm not able to use gpioero 2.1

As previous error reported, I have 1.6.2 and
I did try to install your fix from the provided url but after this i still see 1.6.2 which i think confirms the pre 5 rpis can't use this. Is Physical Button not compatible with gpiozero 1.6.2? I also show python 3.9.2
Thanks