LuxuSam/PhysicalButton #91: [QUESTION] A short question description

ccle610, 2024-08-07:
**Plugin Version**
0.6.2

**Question**
So far I have used a button (GPIO Input) to toggle the light (Action GPIO toggle). 

The output was also used by the Octorelay plug-in to control the light. This worked well for a long time. Unfortunately, the button no longer reacts as long as the Octo Relay plug-in is running. Without the Octorelay plug-in, everything works perfectly!

Is this a topic for here, or for Octo relay?

**Additional context**
OctoRelay Version 5.0.1

ccle610, 2024-08-07:
The solution i found here:
https://github.com/borisbu/OctoRelay/issues/263#issuecomment-2241546096

ccle610, 2024-08-08:
Hi there, sorry for coming back . . . 
I've followed up your hint using tha API of Octorelais, but for me, it does not work outside CLI.
In other words: This Command works well if it is pasted inside a Terminal Session, but don't work inside the Physical Button App:

pi@raspberrypi:~ $ curl 'http://localhost/api/plugin/octorelay'   -H 'X-Api-Key: ****'   -H 'Content-Type: application/json'   -X POST   -d '{ "command": "update", "subject": "r1" }'
{
  "status": true
}

Any Idea, what's going wrong here?
Thx in advance . . . Hans.

LuxuSam, 2024-08-12:
Hi,
Could you check what version of python you are running? Additionally check the version of gpiozero. Maybe there is a version mismatch.

You can can do that by using the following commands in the terminal of the raspberry pi:
- ``` python3 -version ```
- ``` pip3 show gpiozero ```

Cheers,
Sam

LuxuSam, 2024-08-13:
Hi, could you check if the solution in #90 works for you?

ccle610, 2024-08-13:
Hi Sam,
nice to hear from you ;-)
Here are my answers:

Python 3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110] on linux

Name: gpiozero
Version: 1.6.2

When I try to install your new Version, it ends like that:

Successfully uninstalled Physical-Button-0.6.2
Successfully installed Physical-Button-0.6.2 gpiozero-2.0.1 importlib-metadata-4.13.0
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
octoprint-enclosure 4.13.2 requires gpiozero==1.6.2, but you have gpiozero 2.0.1 which is incompatible.
[HardwareButton.txt](https://github.com/user-attachments/files/16599480/HardwareButton.txt)

So, new run with octprint-enclosure uninstalled ends up with gpiozero Version: 1.6.2 . . . seems to hang somewhere . . . 
[log2.txt](https://github.com/user-attachments/files/16599482/log2.txt)

Interesting Line in Log2.txt ?
Requirement already satisfied: gpiozero>1.6.2 in /home/pi/oprint/lib/python3.9/site-packages (from Physical-Button==0.6.2) (2.0.1)

ccle610, 2024-08-13:
OK, I solved it in a different Way ;-)

I've worte a bash script and call this out of your PlugIn - and voila: It works !!!
If someone else is interested:

Inside the Plugin it is a system call:
/home/pi/scripts/light.sh

The bash light.sh

#!/bin/sh
curl 'http://localhost/api/plugin/octorelay' \
  -H 'X-Api-Key: ****verysecretkey****' \
  -H 'Content-Type: application/json' \
  -X POST \
  -d '{ "command": "update", "subject": "r1" }'

Thx for bringing me to this Idea and for publish this Addon, Sam.
Cheers, Hans

ccle610, 2024-08-13:
OK, I solved it in a different Way ;-)

I've worte a bash script and call this out of your PlugIn - and voila: It works !!!
If someone else is interested:

Inside the Plugin it is a system call:
/home/pi/scripts/light.sh

The bash light.sh

#!/bin/sh
curl 'http://localhost/api/plugin/octorelay'
-H 'X-Api-Key: verysecretkey'
-H 'Content-Type: application/json'
-X POST
-d '{ "command": "update", "subject": "r1" }'

Thx for bringing me to this Idea and for publish this Addon, Sam.
Cheers, Hans

closed 2024-08-13 by ccle610

LuxuSam, 2024-08-13:
Hi Hans, great to hear that you were able to solve your problem!

The logs you sent me seem to indicate that gpiozero 2.0.1 was installed, though 😅. Maybe pip will return the correct version after a restart of the pi.