LuxuSam/PhysicalButton #95: [BUG] troubleshooting for older octopi installation - manually installing RPi.GPIO

depperson, 2025-02-14:
**Plugin Version**
https://github.com/LuxuSam/PhysicalButton/archive/refs/heads/82-bug-on-new-installation.zip

**Describe the bug**
I could not get any physical button combination to work. I could confirm that my buttons were configured correctly by using CLI tools on the RPi like `raspi-gpio get 24`.

I'm sorry for going off the rails at this point but it was kind of a ride. 

TL; DR: get inside the octoprint venv and install 
```
cd oprint
source bin/activate
pip3 install RPi.GPIO
```

1. At first my version of gpiozero was incorrect (1.6.2), just like in #90. I installed your .zip plugin fix for that to get the correct version (2.0.1).
2. Still nothing worked. So then I enabled debug logging for your plugin as you described in https://github.com/LuxuSam/PhysicalButton/issues/90#issuecomment-2293088145. That was very helpful. I could see that the plugin was starting up but not showing any activity. Scrolling up a bit in the startup process, I could see a wall of warnings from gpiozero about the pin factories not being installed.
```
2025-02-14 05:12:32,644 - py.warnings - WARNING - /home/pi/oprint/lib/python3.9/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  warnings.warn(

2025-02-14 05:12:32,683 - py.warnings - WARNING - /home/pi/oprint/lib/python3.9/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  warnings.warn(

2025-02-14 05:12:32,711 - py.warnings - WARNING - /home/pi/oprint/lib/python3.9/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  warnings.warn(

2025-02-14 05:12:32,898 - py.warnings - WARNING - /home/pi/oprint/lib/python3.9/site-packages/gpiozero/devices.py:297: NativePinFactoryFallback: Falling back to the experimental pin factory NativeFactory because no other pin factory could be loaded. For best results, install RPi.GPIO or pigpio. See https://gpiozero.readthedocs.io/en/stable/api_pins.html for more information.
  warnings.warn(NativePinFactoryFallback(native_fallback_message))
```
3. I tried installing lgpio but the next startup produced a different error. This is probably due to the ancient OS on my Pi so I did not try fixing it. I only know enough not to fool with GLIBC. 
```
2025-02-14 05:14:30,394 - py.warnings - WARNING - /home/pi/oprint/lib/python3.9/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /home/pi/oprint/lib/python3.9/site-packages/_lgpio.cpython-39-arm-linux-gnueabihf.so)
  warnings.warn(
```
4. Next I tried pigpio. That gave a new error. I guess this library wants to have a daemon running. I didn't want to fix that either. 
```
2025-02-14 05:16:10,763 - py.warnings - WARNING - /home/pi/oprint/lib/python3.9/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from pigpio: failed to connect to localhost:8888
  warnings.warn(
```
5. I finally found the right command to install RPi.GPIO in a way that OctoPi can use it. You cannot install this just as the pi user or system wide, it has to live inside octoprint's virtualenv. I put the commands at the beginning of this rant to fix that. 
6. Now my buttons work!!

If your plugin could install or warn that RPi.GPIO is missing that might help save someone some trouble in the future.

BoilingFrog73, 2025-03-29:
Hi Depperson. I followed the instructions to upgrade gpiozero but still see 1.6.2 reported. 
Is RPi 3 incompatible with later gpiozero version? 
I also did the rpigpio install you mention but still no joy when using button and Physical Button plugin. 
Any help, gratefully received

depperson, 2025-03-29:
If I was to boil down this post into two main points, they would be:
1. Any python libraries need to be installed _inside_ octoprint's venv (virtualenv)
2. The debug logging is a big help and you should use it

These steps worked for me with this hardware combination:
- Raspberry Pi 4 Model B Rev 1.2
- Raspbian GNU/Linux 11 (bullseye)
- OctoPrint Version 1.9.3
- [PhysicalButton plugin with gpiozero fix](https://github.com/LuxuSam/PhysicalButton/archive/refs/heads/82-bug-on-new-installation.zip)

depperson, 2025-03-29:
It is not just a cd command; you must also `activate` the virtual environment (the source command). Your prompt should have an added `(oprint)` prefix while you're "inside" the virtual environment. This makes sure that the pip command installs the python libraries inside octoprint, not system-wide. 
```
pi@octopi:~ $ cd oprint/
pi@octopi:~/oprint $ source bin/activate
(oprint) pi@octopi:~/oprint $ pip3 install RPi-GPIO
```

If you still have trouble after this, I suggest following the [debug instructions](https://github.com/LuxuSam/PhysicalButton/issues/90#issuecomment-2293088145) and file a new issue with the resulting log files.

BoilingFrog73, 2025-03-30:
> (oprint) ***@***.***:~/oprint $ pip3 install RPi-GPIO

That did the trick. Thank you so much.