LuxuSam/PhysicalButton #100: New Octopi kernal has broken RPi.GPIO library, is this plugin still being maintained?

Ecks32, 2025-07-12:
I could not get physical buttons to work with any plugin that supports them.  Thoms who maintains the OctoLight plugin figured out the problem was the new Octopi kernal not working with the RPI.GPIO library.  So no one on the latest version of Octopi is going to be able to use this plugin either.

If this plugin is no longer being maintained then it needs to be removed from Octopi listing.  It's been over a week since I posted my bug report for this plugin and I haven't received any replies.  I'll watch this topic for another week or two and if I don't hear back from someone I'll let Octoprint know that this plugin has been abandoned.  

If you are still active please let me know.  Thanks

Ecks32, 2025-07-14:
Hopefully there is still someone maintaining Physical Buttons so we can keep it around.  

I have never had a printer with an enclosure before, but since Elegoo is having problems manufacturing their new Centuri printers, and offered those of us waiting an upgrade deal, I'm about to get one.  I'll be checking out the Enclosure plugin.  Thanks for the info.

Ohmarinus, 2025-07-14:
> Hopefully there is still someone maintaining Physical Buttons so we can keep it around.
>
> I have never had a printer with an enclosure before, but since Elegoo is having problems manufacturing their [...]

I just went to check the other issues and managed to fix it by combining a couple of fixes that can be found in the issues. 

I fixed it with this:
https://github.com/LuxuSam/PhysicalButton/issues/92#issuecomment-2623462297
(don't forget to install lgpio in the specific env! this is not specifically mentioned in the issue)

and this:
https://github.com/LuxuSam/PhysicalButton/issues/95

Ecks32, 2025-07-14:
That looks like a lot of advanced stuff to a newb.  I may wait until an official fix comes out.   Although, if I get some time I may give it a try.  Thanks for the info.